xmos/fwk_rtos #1: Board support for XVF3600 Eval Kit

mbanth, 2022-05-19:


xmos-jmccarthy, 2022-05-19:
@mbanth Are the Specifications list on the JGB0084 the only peripheral IO required?

mbanth, 2022-05-20:
@xmos-jmccarthy, the specification list at [XK-XVF3600-EVK board](https://xmosjira.atlassian.net/wiki/spaces/BOAR/pages/3097788435/JGB0084) is incomplete.  I suspect that it lists the deltas compared to some other board design that @XMOS-JoeG used as a starting point for this one.  The Port Map spreadsheet on the same page provides further information about the peripheral IO.  However, even it does not make crystal clear that this board supports both I2s and USB IO depending on the firmware in use.

mbanth, 2022-05-24:
I have requested that @tomblackie-xmos send two or three [XK-XVF3600-EVK boards](https://xmosjira.atlassian.net/wiki/spaces/BOAR/pages/3097788435/JGB0084) to the NH office.

keithm-xmos, 2023-03-28:
Closing.  Will re-open if needed.

closed 2023-03-28 by keithm-xmos