xmos/fwk_rtos #2: Add support for Percepio Tracealyzer

keithm-xmos, 2022-05-19:
**Required**
Populate a port layer, most likely to xscope, to handle input and outputs.  

**Nice-to-Have**
There will most likely be some traces that we need to add in to our existing code base as well.  Will likely need to put some unique trace into every ISR call so the tool knows we are in an ISR.
Additionally, we can probably have a trace serve that reads in the trace data over xSCOPE and outputs it over TCP to Tracealyzer.

xmos-jmccarthy, 2022-05-23:
Status update from Percepio: They have worked through tracing issues and are running most of the SMP demos.  Additionally, they have resolved their xscope issues.  They are expecting to be able to release something for us to try again soon.

xmos-jmccarthy, 2022-08-03:
Latest update has been evaluated.  Appears to work as expected now.  Working through how their side is released to see what the SDK will include.

Also looking to purchase a few licenses for internal use.

Closing issue and making new issue in xcore_sdk to integrate Tracealyzer into one of the example applications.

Previous contact will be leaving.  New contact is:
erik.tamlin@percepio.com

closed 2022-08-03 by xmos-jmccarthy