xmos/fwk_rtos #46: Update rtos_qspi_flash driver to use quadflash.h and/or quadflashlib.h.

keithm-xmos, 2022-08-19:
Most should be straight forward. The safety of rtos_qspi_flash_read_ll() needs to be evaluated once implementation is in, but it appears safe from quick glance.

See https://github.com/xmos/xcore_sdk/issues/501

xmos-jmccarthy, 2022-09-22:
Driver updated and verified with #505.

Tools API update requests can be found in: Bug 18717

Update could not be done backwards compatible, so PR will be on hold until XTC 15.2.0 is public, as this tools version will be the required minimum version.

xmos-jmccarthy, 2022-09-26:
https://github.com/xmos/fwk_rtos/tree/feature/tools_qspi_lib
Passes tests with sdk:
https://github.com/xmos/xcore_sdk/tree/feature/tools_qspi_lib

Blocked until public Tools 15.2.0 release.